Tlf/tlf #466: AUTO CQ msg modification

9a3jh, 2025-02-12:
First keyboard key press will stop AUTO CQ message(F12). Now, if someone like to change keyer speed(Pg up/Pg dwn) or AUTO CQ delay(ALt Pg up/Pg dwn), this message will be interrupted.

df7cb, 2025-02-12:
This has also annoyed me in the past... I think AUTO_CQ should even keep sending while I'm trying to edit the last QSO. (Perhaps only the timer should be cancelled, but not the current message being sent.)

N0NB, 2025-02-12:
* On 2025 12 Feb 08:12 -0600, 9a3jh wrote:
> 9a3jh created an issue (Tlf/tlf#466)
> 
> First keyboard key press will stop AUTO CQ message(F12). Now, if
> someone like to change keyer speed(Pg up/Pg dwn) or AUTO CQ delay(ALt
> Pg up/Pg dwn), this message will be interrupted.

It is something I've learned to live with.  Sending other macros are not
affected by a speed change.  Perhaps the F12 behavior could be modified
to only be canceled by key characters that are valid for the call input
field (letters, numerals, and /) and <Escape>.

73, Nate, N0NB

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

dl1jbe, 2025-02-12:
I see the point of changing CW speed and Auto CQ-delay. That should be enabled. It goes along the idea named by Nate.

The immediate stop of transmit was made intentionally on request of some high rate contesters. Together with early start feature it allows to just start typing in a heard call to start a qso. That way there are two key presses saved (ESC to stop Auto CQ and Enter to start sending the others call sign).